ROBOTIS-GIT/ROBOTIS-OP3 #127: error saat menyalakan power pada servo dinamixel

YokeMikal, 2024-02-12:
sayaa baru saja mencoba untuk mengunggah program dinamixel (usb_to_dxl) ke opencr dan setelah tersambung semua ke dinamixel, saat mencoba mengecek program di action editor mendapati error yaitu gagal menyalakan power pada dxls. seperti gambar dibawah ini

[image]

apakah ada solusinya?

shiba-8ro, 2024-02-23:
@YokeMikal 
It looks like an error message saying that the Dynamixel XM430 is not powered on.
In other words, after connecting Dynamixel to OpenCR1.0, you need to turn on the power switch of OpenCR1.0 and confirm that the Dynamixel's pilot LED lights up for 1 second and has booted.

In my experience, when trying out a new firmware, it's a good idea to connect and test only one Dynamixel. Especially if you don't have U2D2.

ROBOTIS-zerom, 2025-03-20:
Due to the lack of a response for an extended period, we will be closing this issue. You can always reopen it if any further questions or problems arise.

closed 2025-03-20 by ROBOTIS-zerom