ROBOTIS-GIT/ROBOTIS-OP3 #134: 3D Model files?

rgamage, 2025-04-08:
I'm looking for any 3D model files (STL, STEP, Blender, etc.) for the OP2 or OP3 robots, which say they are open designs.  I have been unable to find any of these files online.  Any pointer or guidance would be greatly appreciated.  I tried downloading the 4Gb .zip from SourceForge, and I looked in these Github repos, with no success.

Thanks,
Randy

shiba-8ro, 2025-04-09:
@rgamage 
Are the following links helpful to you?
https://github.com/ROBOTIS-GIT/ROBOTIS-OP-Series-Data/tree/master/ROBOTIS-OP3/Hardware/Mechanics/Part
[image]

rgamage, 2025-04-09:
Yes, thank you for that reference.  It was not easy for me to find these files based on repo names and folder/file names.  The next thing I am looking for that I cannot find anywhere is the make/model of the servos / actuators used?  I see the brackets and skins, and the electronic controller (Open CR), but nothing about the servos / actuators themselves, in the manual or other files.

shiba-8ro, 2025-04-10:
I don't know any more information than what's written in the OP3 e-manual.
https://emanual.robotis.com/docs/en/platform/op3/introduction/
I have cloned the OP3 before, and it was a very challenging project. It's important to note that the OP3 hardware is not fully open source.

rgamage, 2025-04-10:
Thank you, I see now it show the actuators as using MX-28 or XM430-W350 DYNAMIXEL proprietary actuators.  I guess this is the part that is not open source.  That's unfortunate, as that seems to be the most expensive part of this kit and therefore limits the ability for anyone to make their own version. I appreciate all the work that went into designing and sharing all the other aspects of this project, which seems to have a very solid architecture and design.  I may use it as a reference for my own, smaller/cheaper version of a walking robot.  Thanks again for your help.